coolio986/Marlin_2.0.x_Anycubic_Chiron #1: Bed Leveling

elstak80, 2020-01-10:
### Description

Bed Leveling is not working

### Steps to Reproduce

1. Upload the firmware
2. Load default setting from TFT
3. Restart the printer
4. pre-heat bed
5. Special Menu--->start bed leveling

**Expected behavior:** [What you expect to happen]
The nozzle home and move to the 1st point
**Actual behavior:** [What actually happens]
No homing and no move of the nozzle, the tft still working and i can move up and down Z. But the leveling procedure not start
#### Additional Information
I just donwload the firmware and compile it using arduino 1.8.10 and TMCstepper library for my TCM2208V3. All other things work, homing from tft... preheat.. filament load... i also try to prehet the nozzle and select "filament change pause" and it work as expected.

coolio986, 2020-01-14:
@elstak80, Due to severe changes in code between marlin 1.1.9 and marlin 2.0, bed leveling is broken. The anycubic way of bed leveling is entirely custom and was never ported to the trunk of marlin. This can be fixed after many code changes, but it will take some time.

coolio986, 2020-01-14:
@elstak80, I have implemented the new version of bed leveling into the latest version. The grid positions don't line up exactly with the grid on the ultrabase. Also, "Advance Setting" doesn't yet work. But I will try and get it working for the next build.

elstak80, 2020-01-15:
Great, just the time to try it an give you the feedback

elstak80, 2020-01-20:
Hi, work quite good, also without the adv. feature. I'm a bit worried about the small amount of free memory available but i had print for 10hrs with no troubles. Next print will be 22hrs, i will update but we can close the issue about leveling

closed 2020-02-11 by coolio986