coolio986/Marlin_2.0.x_Anycubic_Chiron #2: Z-Offset / Probe-Offset / Home

elstak80, 2020-02-07:
### Description

Impossible to adapt the Z value to the current bed high

### Steps to Reproduce

1. Level the bed
2. Start print

**Expected behavior:** 
In the marlin 1.x for the chiron (original or custom) i can set the Z probe offset to compensate the distance between Z=0 and the real bed's positon. Like from the TFT as adv.probing but i can set it inside the fw or using the M581.

**Actual behavior:** 

Set the Z probe offset like in the 1.x.x is not possibile. I tried to set it on the firmware and using the M581 by console, i try negative and positive value but the printer stay always at Z=0 (as per endstops triggering). 

I want to keep the endstops position, alignment will be a pain... I don't need it in the tft but i want to know how to compensate it. from -14 i need -16.6. I Also try to set the home offset M206 but still not working.

elstak80, 2020-02-11:
I see NO update on the tree, last commit on jan.14th 2020, anyway i'm downloading the files again, i'll try again

elstak80, 2020-02-11:
Not working

Ispent few hrs trying to solve...

#define Z_PROBE_LOW_POINT          -5     

That solve the probe not touching the bed

i tried to increase the softwre endstop to -20 and also i tried to disable but when the print start it just stay at Z0 as per optical endstop. still trying...

Biohazzy, 2020-03-15:
Unfortunately I have exactly the same issue. Did you manage to solve the Problem?

coolio986, 2020-04-06:
I am using this exact firmware without any changes on my chiron and I am not having any issues like you describe. First, M581 is not a valid command for Marlin 2.0, M851 is.

After flashing I run commands:
M851 Z-16.70 (mine is 16.70)
G28
G29

Changing Z_PROBE_LOW_POINT to -5 seems risky. If the bed is that badly tilted or wavy then you should contact warranty support.

Try again with those gcode commands and see if it works better.

chiron5498, 2020-09-20:
coolio986 is the stock z probe supposed to work, when i activate it is just stays in the home position and the lcd says probing but it doesnt.  im also running a hermes and my z offset is around z-19.3 . when it homes it stays 6mm above the bed so can do manual level or use the mesh bed level in this firmware . any ideas thx

samuf92, 2020-09-29:
> 
> 
> coolio986 is the stock z probe supposed to work, when i activate it is just stays in the home position and the lcd says probing but it doesnt.

I have exactly the same issue. I'm interested if there any way to use the default chiron probe tool.

elstak80, 2020-09-30:
I download the fw again, flash it, m502, m500, reset, auto_level.gocode, probe.... and it works. I think as is the fw is ok. If you start changing things like the probe grid.... well it fail.

So as i had open the post, coolio can close it.

closed 2020-10-03 by coolio986