coolio986/Marlin_2.0.x_Anycubic_Chiron #10: Error in Arduino IDE

Kjaer79, 2020-09-22:
My Anycubic Chiron is 100% stock.
and not customized in any way.

I download your repo today as ZIP
And have the most recent Arduino IDE (downloadet today to) 

I did follow these instructions: https://github.com/coolio986/Marlin_2.0.x_Anycubic_Chiron#how-to-flash-this
But cant choose "Genuino Mega 2560 and ATmega2560" only "Arduino Mega or Mega 2560"

And get this error.

"Arduino: 1.8.13 (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\src\module/stepper/indirection.h:44:0,

                 from sketch\src\module/stepper.h:47,

                 from sketch\src\Marlin.cpp:37:

sketch\src\module/stepper/trinamic.h:31:4: error: #error "Update TMCStepper library to 0.5.0 or newer."

   #error "Update TMCStepper library to 0.5.0 or newer."

    ^~~~~

In file included from sketch\src\module/stepper/trinamic.h:35:0,

                 from sketch\src\module/stepper/indirection.h:44,

                 from sketch\src\module/stepper.h:47,

                 from sketch\src\Marlin.cpp:37:

c:\users\jakobk~1\appdata\local\temp\arduino_build_883821\sketch\src\feature\tmc_util.h:29:10: fatal error: TMCStepper.h: No such file or directory

 #include <TMCStepper.h>

          ^~~~~~~~~~~~~~

compilation terminated.

exit status 1

Error compiling for board Arduino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences."

What do i do wrong?

samuf92, 2020-09-24:
Hi.

I'm interested in this discussion. I have a similar issue, a lot of errors when I try to verify sketch.

an error is like this:

In file included from sketch\src\module\stepper\trinamic.h:35:0,
                 from sketch\src\module\stepper\trinamic.cpp:32:
c:\users\samuele\appdata\local\temp\arduino_build_591907\sketch\src\feature\tmc_util.h: In instantiation of 'TMCMarlin<TMC2208Stepper, AXIS_LETTER, DRIVER_ID, AXIS_ID>::TMCMarlin(uint16_t, uint16_t, float, uint8_t, bool) [with char AXIS_LETTER = 'X'; char DRIVER_ID = '0'; AxisEnum AXIS_ID = (AxisEnum)0; uint16_t = unsigned int; uint8_t = unsigned char]':
sketch\src\module\stepper\trinamic.cpp:207:7:   required from here
c:\users\samuele\appdata\local\temp\arduino_build_591907\sketch\src\feature\tmc_util.h:175:40: warning: 'TMC2208Stepper::TMC2208Stepper(uint16_t, uint16_t, float, bool)' is deprecated: Boolean argument has been deprecated and does nothing [-Wdeprecated-declarations]
       TMC2208Stepper(RX, TX, RS, has_rx)

and an infinite series of errors like this:

In file included from c:\users\samuele\appdata\local\temp\arduino_build_591907\sketch\src\pins\pins.h:162:0,
                 from c:\users\samuele\appdata\local\temp\arduino_build_591907\sketch\src\inc\marlinconfig.h:32,
                 from c:\users\samuele\appdata\local\temp\arduino_build_591907\sketch\src\gcode\gcode.h:282,
                 from sketch\src\gcode\temperature\M105.cpp:23:
c:\users\samuele\appdata\local\temp\arduino_build_591907\sketch\src\pins\ramps/pins_TRIGORILLA_14.h:146:0: warning: "HEATER_BED_PIN" redefined
   #define HEATER_BED_PIN   45 // RAMPS_D8_PIN

Can someone help me to fix that, i'm really interested to try this firware on my printer.

kraloulk, 2020-09-28:
Use visual studio code to install it to your printer. Few good walkthroughs on YouTube for that.

samuf92, 2020-09-29:
> 
> 
> Use visual studio code to install it to your printer. Few good walkthroughs on YouTube for that.

It's work for me, thanks. It found some errors during compiling but at the end it says success and the printer works without issue.

coolio986, 2020-09-30:
Since this version of marlin contains 32bit libraries, there are some issues compiling that in the latest arduino version. As @kraloulk mentioned its best to use visual studio code for this since it does a better job handling this. At some point I will update this repository with the VS walk through unless someone wants to make one.

raindancer2204, 2021-01-02:
I will now also try Visual Studio because I came across the same error as samuf92.

Short question can someone update the documentation on this? Unfortunately I don't have write access :) otherwise I would support.