coolio986/Marlin_2.0.x_Anycubic_Chiron #12: Cant level below Z_Endstops

cryoplay, 2020-10-12:
Hi there.

I have a massive modded Anycubic Chiron.
I Upgraded all Axes with Linear Rails including the Printbed.
Its now fixed without springs.

Also my hotend is a watercooled direct BMG hotend.

So i wanted to try your Firmware.

But when i want to level the bed i cant go below the Z_Endstops.

My Printbed is about 0.35 below the Z Endstops.

Can someone help me with this Issue?

alexnv, 2020-10-22:
i'm have same problem

TheDezzoSide, 2021-02-05:
Same problem!!

alantum, 2022-01-30:
Same problem

eXcalibre, 2023-12-12:
Ok so this is not a direct answer to the original poster (sorry), but hopefully it may help if you still have the adjustment springs on the print bed.

I had the same issue when attempting to set up mine.
I needed to reverse what i had always been told (_release the bed adjusters to be as far away from the print head_).
So **INSTEAD**, raise the bed a little, but bear in mind that the head WILL interfer until you amend the level with the hidden menu alignment.

So i needed to use the bed springs to raise the bed and then use the firmware to level the bed afterwards. This gave me the ability to get my Z axis leveled.

Amand1100, 2025-08-12:
The problem is, even if you disable the Z endstop marlin blocks you from going below z = 0 after a few points. You have to level your bad manually with g code input via pronterface or similar. Chatgpt gave me the GCode to reach all the points to measure the offset there. Secure all the measured offsets for each points in an excel document to be safe. And do M500 and G28 After a few measured points so the added up error doesn't get too high. After zeroing with G28 you can just directly go to the point which would come next.

After doing G91 lower the nozzle to the bed. Remember that value and drive that value the same amount up again. Then do G90 and Safe the value with M421 I Y (of the point) Z(Value).

 Here ist the list of Gcode made by Chatgpt: 

; Anycubic Chiron 5x5 Mesh Leveling - Komplettliste mit 90°C Bett und 238°C Hotend
M211 S0              ; Softendstops ausschalten
G28                  ; Home alle Achsen
M140 S90             ; Bett auf 90°C vorheizen
M190 S90             ; Warten bis Bett 90°C erreicht
M104 S238            ; Hotend auf 238°C vorheizen
M109 S238            ; Warten bis Hotend 238°C erreicht
G0 Z10 F6000         ; Z auf sichere Höhe fahren

; Punkte 5x5 (I=0..4, J=0..4), Abstand 90mm, Start bei 20mm

; === J=0 (vorne) ===
G0 X20  Y20  F6000
G91
; Z manuell justieren (z.B. G0 Z-0.05)
G90
M421 I0 J0 Z-0.00

G0 X110 Y20  F6000
G91
; Z manuell justieren
G90
M421 I1 J0 Z-0.00

G0 X200 Y20  F6000
G91
; Z manuell justieren
G90
M421 I2 J0 Z-0.00

G0 X290 Y20  F6000
G91
; Z manuell justieren
G90
M421 I3 J0 Z-0.00

G0 X380 Y20  F6000
G91
; Z manuell justieren
G90
M421 I4 J0 Z-0.00

; === J=1 ===
G0 X20  Y110 F6000
G91
; Z manuell justieren
G90
M421 I0 J1 Z-0.00

G0 X110 Y110 F6000
G91
; Z manuell justieren
G90
M421 I1 J1 Z-0.00

G0 X200 Y110 F6000
G91
; Z manuell justieren
G90
M421 I2 J1 Z-0.00

G0 X290 Y110 F6000
G91
; Z manuell justieren
G90
M421 I3 J1 Z-0.00

G0 X380 Y110 F6000
G91
; Z manuell justieren
G90
M421 I4 J1 Z-0.00

; === J=2 ===
G0 X20  Y200 F6000
G91
; Z manuell justieren
G90
M421 I0 J2 Z-0.00

G0 X110 Y200 F6000
G91
; Z manuell justieren
G90
M421 I1 J2 Z-0.00

G0 X200 Y200 F6000
G91
; Z manuell justieren
G90
M421 I2 J2 Z-0.00

G0 X290 Y200 F6000
G91
; Z manuell justieren
G90
M421 I3 J2 Z-0.00

G0 X380 Y200 F6000
G91
; Z manuell justieren
G90
M421 I4 J2 Z-0.00

; === J=3 ===
G0 X20  Y290 F6000
G91
; Z manuell justieren
G90
M421 I0 J3 Z-0.00

G0 X110 Y290 F6000
G91
; Z manuell justieren
G90
M421 I1 J3 Z-0.00

G0 X200 Y290 F6000
G91
; Z manuell justieren
G90
M421 I2 J3 Z-0.00

G0 X290 Y290 F6000
G91
; Z manuell justieren
G90
M421 I3 J3 Z-0.00

G0 X380 Y290 F6000
G91
; Z manuell justieren
G90
M421 I4 J3 Z-0.00

; === J=4 (hinten) ===
G0 X20  Y380 F6000
G91
; Z manuell justieren
G90
M421 I0 J4 Z-0.00

G0 X110 Y380 F6000
G91
; Z manuell justieren
G90
M421 I1 J4 Z-0.00

G0 X200 Y380 F6000
G91
; Z manuell justieren
G90
M421 I2 J4 Z-0.00

G0 X290 Y380 F6000
G91
; Z manuell justieren
G90
M421 I3 J4 Z-0.00

G0 X380 Y380 F6000
G91
; Z manuell justieren
G90
M421 I4 J4 Z-0.00

; Mesh speichern & Softendstops wieder aktivieren
M500
M211 S1