coolio986/Marlin_2.0.x_Anycubic_Chiron #13: visual studio code issue

Metzis, 2020-11-19:
hey i could use some help with this one. I went with vsc cause i read that is more flexible with this mode that arduino  but still got an error! i made my changes i config.h saved it and tried to build but i got this! please help!
by the way thanks coolio986 for the hard work here!
[image]

coolio986, 2020-11-19:
Can you post your config and config_adv?

Metzis, 2020-11-20:
coolio986 i feel stupid right now. I was right on patching the files to send them over and i went over the code one more time! Guess what, i had the port value wrong :P change it back to 0 and everything is golden! thankS and sorry for the waste of time!

closed 2020-11-20 by Metzis